SmartArduino/SZDOITWiKi #2: Does ESP-S have bottom pins or wrong image uploaded on wiki?

humperdink001, 2017-05-01:
I have been looking to find an ESP-12 without the new 6 pins on the bottom of the board.
Your description of ESP-S on the wiki page says
 "In essense, ESP-S is very similar to ESP-F. The only difference is that ESP-F has more 6 solder point in the end than ESP-S. Therefore, in many applications. This Two WiFi modules can be changed each other."

But the picture and dimensions image on the ESP-S wiki page and the ESP-S pdf manual show the 6 bottom pins. Does the ESP-S have the bottom pins or did you accidentally place a wrong image of ESP-F on this pages.
https://github.com/SmartArduino/SZDOITWiKi/wiki/ESP8266---ESP-S
https://github.com/SmartArduino/ESP/blob/master/UserManualForESP-S_doitV1.pdf

SmartArduino, 2017-05-02:
yes, hi guys, do you need esp-s or esp-f, esp-s has no 6 pins on the board. and the board has some problem

With the best regards. Any problem, please don't hesitate to contact us. Thanks again.

Yours Sincerely, Respectfully and Gratefully,
yichone (Ph.D.)

Our website:
Shop site: www.smartarduino.com 
Official site: www.doit.am 
WiFiMCU: www.wifimcu.com 
Yun Programming: https://github.com/SmartArduino/COULD-coding-with-NodeMCU 

New products
SP8266: http://www.smartarduino.com/the-newest-v3-nodemcu-based-on-esp-12e-from-esp8266_p94571.html 
WiFi Shield for Arduino: http://www.smartarduino.com/esp8266-esp-13-wifi-web-sever-shield-for-arduino_p94660.html 
Esp8266 Development kit: http://www.smartarduino.com/nodemcu-development-kit-nodemcu-motor-shield-l293dd_p94867.html 
WiFiMCU: http://www.smartarduino.com/wifimcu-wireless-wifi-development-board-using-lua-from_p94744.html 
Raspberry pi Camera: http://www.smartarduino.com/rpi-cameras_d001003003.html
Tank Chassis:
http://www..smartarduino.com/t200-aluminum-alloy-metal-tank-track-caterpillar-chassis_p94742.html 
http://www.smartarduino.com/t100-aluminum-alloy-metal-tank-track-caterpillar-chassis_p94741.html 
http://www.smartarduino.com/t300-aluminum-alloy-metal-tank-track-caterpillar-chassis_p94740.html 
http://www.smartarduino.com/c300-aluminum-alloy-metal-wheel-car-chassis-for-arduino_p94743.html 

Contact Information
Company Name: Shenzhen Doctors of Intelligence & Technology (Shenzhen DOIT)
Addr.: Room 3305, Lixin Garden, Xili Town, Nanshan Distr., Shenzhen 518052, China
Skype: yichone
WhatsAPP: +8618676662425
Mobile: +86-18676662425
Emails: service@smartarduino.com 
          yichoneyi@163.com 
          yichone@doit.am 
 
From: humperdink001
Date: 2017-05-02 02:05
To: SmartArduino/SZDOITWiKi
CC: Subscribed
Subject: [SmartArduino/SZDOITWiKi] Does ESP-S have bottom pins or wrong image uploaded on wiki? (#2)
I have been looking to find an ESP-12 without the new 6 pins on the bottom of the board.
Your description of ESP-S on the wiki page says
"In essense, ESP-S is very similar to ESP-F. The only difference is that ESP-F has more 6 solder point in the end than ESP-S. Therefore, in many applications. This Two WiFi modules can be changed each other."
But the picture and dimensions image on the ESP-S wiki page and the ESP-S pdf manual show the 6 bottom pins. Does the ESP-S have the bottom pins or did you accidentally place a wrong image of ESP-F on this pages.
https://github.com/SmartArduino/SZDOITWiKi/wiki/ESP8266---ESP-S
https://github.com/SmartArduino/ESP/blob/master/UserManualForESP-S_doitV1.pdf

SmartArduino, 2017-05-02:
Thanks, already changed

https://github.com/SmartArduino/SZDOITWiKi/wiki/ESP8266---ESP-S 

With the best regards. Any problem, please don't hesitate to contact us. Thanks again.

Yours Sincerely, Respectfully and Gratefully,
yichone (Ph.D.)

Our website:
Shop site: www.smartarduino.com 
Official site: www.doit.am 
WiFiMCU: www.wifimcu.com 
Yun Programming: https://github.com/SmartArduino/COULD-coding-with-NodeMCU 

New products
SP8266: http://www.smartarduino.com/the-newest-v3-nodemcu-based-on-esp-12e-from-esp8266_p94571.html 
WiFi Shield for Arduino: http://www.smartarduino.com/esp8266-esp-13-wifi-web-sever-shield-for-arduino_p94660.html 
Esp8266 Development kit: http://www.smartarduino.com/nodemcu-development-kit-nodemcu-motor-shield-l293dd_p94867.html 
WiFiMCU: http://www.smartarduino.com/wifimcu-wireless-wifi-development-board-using-lua-from_p94744.html 
Raspberry pi Camera: http://www.smartarduino.com/rpi-cameras_d001003003.html
Tank Chassis:
http://www..smartarduino.com/t200-aluminum-alloy-metal-tank-track-caterpillar-chassis_p94742.html 
http://www.smartarduino.com/t100-aluminum-alloy-metal-tank-track-caterpillar-chassis_p94741.html 
http://www.smartarduino.com/t300-aluminum-alloy-metal-tank-track-caterpillar-chassis_p94740.html 
http://www.smartarduino.com/c300-aluminum-alloy-metal-wheel-car-chassis-for-arduino_p94743.html 

Contact Information
Company Name: Shenzhen Doctors of Intelligence & Technology (Shenzhen DOIT)
Addr.: Room 3305, Lixin Garden, Xili Town, Nanshan Distr., Shenzhen 518052, China
Skype: yichone
WhatsAPP: +8618676662425
Mobile: +86-18676662425
Emails: service@smartarduino.com 
          yichoneyi@163.com 
          yichone@doit.am 
 
From: humperdink001
Date: 2017-05-02 02:05
To: SmartArduino/SZDOITWiKi
CC: Subscribed
Subject: [SmartArduino/SZDOITWiKi] Does ESP-S have bottom pins or wrong image uploaded on wiki? (#2)
I have been looking to find an ESP-12 without the new 6 pins on the bottom of the board.
Your description of ESP-S on the wiki page says
"In essense, ESP-S is very similar to ESP-F. The only difference is that ESP-F has more 6 solder point in the end than ESP-S. Therefore, in many applications. This Two WiFi modules can be changed each other."
But the picture and dimensions image on the ESP-S wiki page and the ESP-S pdf manual show the 6 bottom pins. Does the ESP-S have the bottom pins or did you accidentally place a wrong image of ESP-F on this pages.
https://github.com/SmartArduino/SZDOITWiKi/wiki/ESP8266---ESP-S
https://github.com/SmartArduino/ESP/blob/master/UserManualForESP-S_doitV1.pdf

humperdink001, 2017-05-02:
Still wrong pictures and measure diagram on the pdf manual https://github.com/SmartArduino/ESP/blob/master/UserManualForESP-S_doitV1.pdf

Also on the wiki https://github.com/SmartArduino/SZDOITWiKi/wiki/ESP8266---ESP-S
I see you change to the correct photo jpg of ESP-S but the measure diagram still is showing the bottom 6 pins which might still confuse other people.
I would recommend taking the extra time when you get a chance to create and upload a new image of the measurement diagram without the bottom pins. On the pdf manual page also.
--------------------------------------------

 

 
 
 
 
 
 https://github.com/SmartArduino/SZDOITWiKi/wiki/ESP8266---ESP-S
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 https://github.com/SmartArduino/COULD-coding-with-NodeMCU 
 
 
 
 
 
 
 
 
 
 
 http://www.smartarduino.com/the-newest-v3-nodemcu-based-on-esp-12e-from-esp8266_p94571.html
 
 
 
 http://www.smartarduino.com/esp8266-esp-13-wifi-web-sever-shield-for-arduino_p94660.html
 
 
 
 http://www.smartarduino.com/nodemcu-development-kit-nodemcu-motor-shield-l293dd_p94867.html
 
 
 
 http://www.smartarduino.com/wifimcu-wireless-wifi-development-board-using-lua-from_p94744.html
 
 
 
 http://www.smartarduino.com/rpi-cameras_d001003003.html
 
 
 
 
 http://www..smartarduino.com/t200-aluminum-alloy-metal-tank-track-caterpillar-chassis_p94742.html
 
 
 
 http://www.smartarduino.com/t100-aluminum-alloy-metal-tank-track-caterpillar-chassis_p94741.html
 
 
 
 http://www.smartarduino.com/t300-aluminum-alloy-metal-tank-track-caterpillar-chassis_p94740.html
 
 
 
 http://www.smartarduino.com/c300-aluminum-alloy-metal-wheel-car-chassis-for-arduino_p94743.html
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
  
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 https://github.com/SmartArduino/SZDOITWiKi/wiki/ESP8266---ESP-S
 
 
 https://github.com/SmartArduino/ESP/blob/master/UserManualForESP-S_doitV1.pdf
 
 
 
 
 
 
 thread.

humperdink001, 2017-05-02:
Thanks for the reply. What do you mean "and the board has some problem"? I ordered april 28,  25 ESP-S boards for testing and if it works for my project I will be ordering 200-500 more.

I look forward to testing with your product. I ordered the M2 and ESP-1 last weekend and paid for 5-8 day Fedex shipping on aliexpress because I am launching new product in few weeks and I think M2 board will work better for my project. My order still has not shipped. Can you help get my order completed and shipped. I ordered some boards from another seller also to test and they already shipped on Monday and I only paid regular shipping with them.

I am attaching pdf of my order.

Thank you very much!!
--------------------------------------------

 

 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 https://github.com/SmartArduino/COULD-coding-with-NodeMCU 
 
 
 
 
 
 
 
 
 
 
 http://www.smartarduino.com/the-newest-v3-nodemcu-based-on-esp-12e-from-esp8266_p94571.html
 
 
 
 http://www.smartarduino.com/esp8266-esp-13-wifi-web-sever-shield-for-arduino_p94660.html
 
 
 
 http://www.smartarduino.com/nodemcu-development-kit-nodemcu-motor-shield-l293dd_p94867.html
 
 
 
 http://www.smartarduino.com/wifimcu-wireless-wifi-development-board-using-lua-from_p94744.html
 
 
 
 http://www.smartarduino.com/rpi-cameras_d001003003.html
 
 
 
 
 http://www..smartarduino.com/t200-aluminum-alloy-metal-tank-track-caterpillar-chassis_p94742.html
 
 
 
 http://www.smartarduino.com/t100-aluminum-alloy-metal-tank-track-caterpillar-chassis_p94741.html
 
 
 
 http://www.smartarduino.com/t300-aluminum-alloy-metal-tank-track-caterpillar-chassis_p94740.html
 
 
 
 http://www.smartarduino.com/c300-aluminum-alloy-metal-wheel-car-chassis-for-arduino_p94743.html
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
  
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 https://github.com/SmartArduino/SZDOITWiKi/wiki/ESP8266---ESP-S
 
 
 https://github.com/SmartArduino/ESP/blob/master/UserManualForESP-S_doitV1.pdf