SmartArduino/SZDOITWiKi #4: No responses to AT+ commands

DazWilkin, 2017-05-22:
I'm able to send values to the TTLtoWiFi and see these reproduced on the serial device connection.

I don't receive responses to the AT commands documented in the manual.

What am I doing wrong?

```
telnet> open 192.168.4.1 9000
Trying 192.168.4.1...
Connected to 192.168.4.1.
Escape character is '^]'.
Test
AT+STAINFO
AT+STASTATUS
Test
```

And, on the serial connection:

```
Test
AT+STAINFO
AT+STASTATUS
Test
```

I expect responses to the AT+ commands but don't appear to be receiving anything.

m-sja, 2017-05-25:
I had the same issue.
Switching to “no line ending“ solved the issue for me.

DazWilkin, 2017-05-25:
@m-sja thank you! I'll try it today

SmartArduino, 2017-05-27:
please try to use the AT command at serial port, Note NET interface.

m-sja, 2017-05-27:
If you send [LF] and or [CR] after the AT-Command the Module passes it through from MCU to NET instead of responding to the Command back to MCU.
I would call this a Bug as this is not documented.

DazWilkin, 2017-05-27:
Thank you both for your help.

I am able successfully to issue AT commands via the *serial* connection suggested by @SmartArduino and using @m-sja suggested "no line ending" (e.g. `AT+STAINFO`, `AT+STASTATUS`) and receive the expected responses.

I am unable to use `screen` and `PuTTY` for the serial comms but this works with the Arduino IDE Serial Monitor.

GeoShu, 2019-12-20:
Hello! I connect the DT-06 through the 2 motor & 16 servo board. When I send the AT + STAINFO command through the Port Monitor, there is no response. The sketch is written in Arduino:
#include <SPI.h>
#include <WiFiServer.h>
#include <WiFiUdp.h>
#include <WiFiClient.h>
#include <WiFi.h>
#include <SoftwareSerial.h>

String inString;
int inChar;

SoftwareSerial portOne (3, 2);

void setup () {
    Serial.begin (9600);
    while (! Serial) {
      ;
    }
    portOne.begin (9600);
}

void loop () {
    portOne.listen ();
    while (portOne.available ()> 0) {
      inChar = portOne.read ();
      inString + = (char) inChar;
      if (inChar == '\ n') {
        Serial.print (inString);
      }
      inString = "";
    }
}

How to make DT-06 execute AT commands?