SmartArduino/SZDOITWiKi #6: Uploading stub... Timed out waiting for packet header (ESP32)

opichals, 2017-06-14:
I am trying to flash this board:
https://www.aliexpress.com/item/ESP-32S-Development-Board-Esp32-Module-2-4-GHz-dual-mode-Wi-Fi-and-Bluetooth-chips/32803195774.html

The esptool has the following trouble:
```
Arduino: 1.8.2 (Mac OS X), Board: "ESP32 Dev Module, 40MHz, 921600, Verbose"

Archiving built core (caching) in: /var/folders/kd/dms1b9n94m3_5fn8rg8s_k7c0000gn/T/arduino_cache_973448/core/core_espressif_esp32_esp32_FlashFreq_40,UploadSpeed_921600,DebugLevel_verbose_284d0d6d1a9f5d714c68e4f94b1da0ff.a
Sketch uses 120768 bytes (9%) of program storage space. Maximum is 1310720 bytes.
Global variables use 9332 bytes (3%) of dynamic memory, leaving 285580 bytes for local variables. Maximum is 294912 bytes.
/Users/standa/Documents/Arduino/hardware/espressif/esp32/tools/esptool --chip esp32 --port /dev/cu.SLAB_USBtoUART --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_freq 40m --flash_mode dio --flash_size 4MB 0x1000 /Users/standa/Documents/Arduino/hardware/espressif/esp32/tools/sdk/bin/bootloader.bin 0x8000 /var/folders/kd/dms1b9n94m3_5fn8rg8s_k7c0000gn/T/arduino_build_102618/BareMinimum.ino.partitions.bin 0xe000 /Users/standa/Documents/Arduino/hardware/espressif/esp32/tools/partitions/boot_app0.bin 0x10000 /var/folders/kd/dms1b9n94m3_5fn8rg8s_k7c0000gn/T/arduino_build_102618/BareMinimum.ino.bin 
esptool.py v2.0-beta3
Connecting....
Uploading stub...

A fatal error occurred: Timed out waiting for packet header
A fatal error occurred: Timed out waiting for packet header
```

I have tried different baud rates as well as flash_freq changes to no help.

opichals, 2017-06-14:
Reportedly there are more users with the same issue https://github.com/espressif/arduino-esp32/issues/409 and the following trouble could be related https://github.com/espruino/Espruino/issues/1177.

PheiPheiPhei, 2017-08-20:
I can confirm this issue on my DoIt, too. Guess I'll be using a different board in the future.

w8an, 2017-09-18:
Removing the ESP32 from the breadboard during programming really does make a difference. I'm using the Doit ESP32 Devkit V1 with the Arduino IDE set on board: ESP32 Dev Module with the Upload Speed baudrate at 115200. If fails with the "A fatal error occurred: Timed out waiting for packet header" when breadboarded, but succeeds when laying on the bench unconnected except for the usb cable. No button presses are required prior to upload.

kosso, 2018-03-16:
Setting the upload speed to 115200 fixed this for me.