SmartArduino/SZDOITWiKi #7: Fritzing file for ESP-32 Development Board

urs8000, 2017-08-24:
It would be very nice & helpful if you could create a parts-file for the (big) ESP-32 Development Board (2 x 15 pins)

anthonysleck, 2018-01-26:
Still nothing on this?

ve7it, 2018-03-13:
a schematic for the new 18 pin *2  esp32 devkit v1 board is REALLY needed to use this board... I am very frustrated at spending all afternoon trying to find out how to use this board pinout.
PLEASE post schematic for product

dimension1991, 2018-07-16:
i also wanted the esp32 2x15 pin fritzing part

efi, 2018-09-09:
There seems to exist something like that:

https://lastminuteengineers.com/esp32-dht11-dht22-web-server-tutorial/#attachment_818

But it is not available from anywhere. Anyone want to contact the people over there? The author seems to be "lmewp1908" - but this name does nt seem to appear anywhere else on the web.

csaraivarocha, 2018-10-15:
Hi everybody!

> i also wanted the esp32 2x15 pin fritzing part
> [image]

Did you find its fritzing part?

efi, 2018-10-15:
The people at lastminuteengineers did not respond to my inquiry...

csaraivarocha, 2018-10-15:
Really?!??! It’s sad. Because I’ve just maken a question  over there.

If I find something I’ll post here.

abs/csr

csaraivarocha, 2018-10-16:
Hello!
I got it!

De: Cristiano S Rocha <cristianosrocha@hotmail.com>
Enviado: Monday, October 15, 2018 5:40:46 PM
Para: SmartArduino/SZDOITWiKi; SmartArduino/SZDOITWiKi
Cc: csaraivarocha; Comment
Assunto: Re: [SmartArduino/SZDOITWiKi] Fritzing file for ESP-32 Development Board (#7)

Really?!??! It’s sad. Because I’ve just maken a question  over there.

If I find something I’ll post here.

abs/csr

efi, 2018-10-18:
Oh, cool. Are you able/allowed to share it (publicly or private)?

csaraivarocha, 2018-10-21:
>           Oh, cool. Are you able/allowed to share it (publicly or private)?

Sorry @efi, I was so happy, but its fritzing part was not designed completed at all. But just the image looks like DoIt DevKit ESP32. :( :(

jakobtholm, 2021-02-11:
Did anybody find this part and if so can they post a link?

joaquinperaza, 2021-06-17:
Hi, good news

Found here:
https://qiita.com/XZ_Manj/items/8bbdef59eeac4286e4f8

Downloaded from here:
https://github.com/Atsumitsu/fritzing-parts