SmartArduino/SZDOITWiKi #23: Related to DT-06 Latest Firmware

dev-ahmed-ibrahim, 2019-07-09:
Hi.

Is there any link to dowmload the latest firmware of Serial to Wi-Fi firmware version 3.

Thanks.

SmartArduino, 2019-07-10:
no, we don't plan to open the newest firmware.

With the best regards
Yichone (Manager)
DOIT Ltd., from Shenzhen

Resources and documents: wiki.doit.am
site: www.doit.am 
shop: www.vvdoit.com, www.smartarduino.com 
whatsapp: +8618676662425
wechat: 18676662425
Emails: yichone@doit.am, yichoneyi@163.com 
 
From: dev-ahmed-ibrahim
Date: 2019-07-09 23:20
To: SmartArduino/SZDOITWiKi
CC: Subscribed
Subject: [SmartArduino/SZDOITWiKi] Related to DT-06 Latest Firmware (#23)
Hi.
Is there any link to dowmload the latest firmware of Serial to Wi-Fi firmware version 3.
Thanks.

athertop, 2019-10-21:
I am not interested in the source code, but just the required bin files to re-flash the doit ttl-wifi board - can you please provide a link? My board needs re-flashing

europacafe, 2021-02-16:
May I have the link to download the DT-06 stock firmware too? I need to reflash my Dt-06. Thank you.

totoantibes, 2021-06-02:
same here. no need for source code but just need to reflash DT-06 with your original stoc firmware

europacafe, 2021-06-02:
I bought another DT-06 for another project and managed to extract it stock firmware out. I did try installing it back to another DT-06 which I put other firmware in, and it worked as stock perfectly.
Here is the link to download it. It's a 4MB bin file.
[https://drive.google.com/file/d/1oJyDso02NVfE8YmhDvKFJmyJGmnOsdw2/view?usp=sharing](url)

totoantibes, 2021-06-02:
than you. It almost worked. I have an MD5 error at the end of the writing procedure...
```

File  md5: ad84ed661996d705f1fa84d0fb1a52f0
Flash md5: 932a85921e773fb01bf27ee8ef352079
MD5 of 0xFF is 2b7a70fa59f8173635bcbe956bad56c6

A fatal error occurred: MD5 of file does not match data in flash!

```

europacafe, 2021-06-02:
[image]
Strange. This was what I did and documented the result.

totoantibes, 2021-06-02:
the only diff i see is that my recognized chip is ESP8285N08

europacafe, 2021-06-03:
I zipped the same bin file and upload to Google Drive. I'm not sure it will make any difference.
[https://drive.google.com/file/d/1Z19SgD2dpa5PM9bHKWDcOLvxZjFnL2d3/view?usp=sharinghttps://drive.google.com/file/d/1Z19SgD2dpa5PM9bHKWDcOLvxZjFnL2d3/view?usp=sharing](url)

totoantibes, 2021-06-08:
thx but i saw that the size of the file is too large. my version of the chip is 1MB of memory.  The one you have is 4MB.

totoantibes, 2021-06-11:
I had one extra dt06 too that i thought i could not get to but in the end i managed to plug it in and read the flash. difference is that i used the following read command: 
`esptool -p COM4 -b 460800 read_flash 0 0x100000 dt06_backup.bin`

Notice the 0x100000 which spits out the 1MB bin required for the DT-05 ESP8285 chip. 

here is the bin. 
https://drive.google.com/file/d/10xS_OjJX_gLRITfhas6bxUiCrY33fH5l/view?usp=sharing

shro70, 2021-07-04:
I tied your 1mb firmware on a esp01 without success.
It flash correctly but nothing after a reboot/cycle power 
No wifi appear and it output strange characters in the serial monitor .

wifixcort, 2023-10-20:
Same here, waiting for the firmware.bin to restore some 8285

hwti, 2023-12-04:
I got 2 DT-06 (clone ?) from Aliexpress, with ESP8285H16 and a very old AT firmware (1.1.0.0).

I found `WIFI_TTL_3.0_191128.bin` on http://bbs.doit.am/forum.php?mod=viewthread&tid=472.
It works, but the SoftAP is always enabled even if I disable it in the web interface.
The web page mentions `© 2014-2017 All right reversed.`, and I also see `Doit_Ser2Sock_3.0_20170320`.
This firmware is based on `OS SDK ver: 2.0-dev(016fc33) compiled @ Jan  5 2018 13:51:41`.

With the saved firmware posted here, I get a boot loop.
The binary contains `© 2014-2018 All right reversed.` and `Doit_Ser2Sock_3.2.1_20171229`, which suggests it might be more recent.
But it is based on an older SDK : `OS SDK ver: 1.5.0-dev(950076a) compiled @ Nov  4 2016 19:29:32`.

SmartArduino, 2023-12-04:
then you get a fake one

Thanks a lot and best regards.
Any problem, please feel free to let me know. Thanks again.

Yours,
CarterYee
SnapChat: itchenve
WhatsApp: 0086-18676662425
Skype: yichone
Emails: ***@***.******@***.***
WeChat: itchenve

Documents: https://gitnova.com/#/
Online shop: http://vvdoit.com/

---- Replied Message ----
| From | Loïc ***@***.***> |
| Date | 12/5/2023 04:59 |
| To | ***@***.***> |
| Cc | ***@***.***> ,
***@***.***> |
| Subject | Re: [SmartArduino/SZDOITWiKi] Related to DT-06 Latest Firmware (#23) |

I got 2 DT-06 (clone ?) from Aliexpress, with ESP8285H16 and a very old AT firmware (1.1.0.0).

I found WIFI_TTL_3.0_191128.bin on http://bbs.doit.am/forum.php?mod=viewthread&tid=472.
It works, but the SoftAP is always enabled even if I disable it in the web interface.
The web page mentions © 2014-2017 All right reversed., and I also see Doit_Ser2Sock_3.0_20170320.
This firmware is based on OS SDK ver: 2.0-dev(016fc33) compiled @ Jan 5 2018 13:51:41.

With the saved firmware posted here, I get a boot loop.
The binary contains © 2014-2018 All right reversed. and Doit_Ser2Sock_3.2.1_20171229, which suggests it might be more recent.
But it is based on an older SDK : OS SDK ver: 1.5.0-dev(950076a) compiled @ Nov 4 2016 19:29:32.